sparkfun/SparkFun_KX13X_Arduino_Library #8: getRawAccelData() issue regarding reading the buffer

ahermida69, 2021-08-30:
The function getRawAccelData()  checks the BUFFER FULL Interrupt to figure out if the buffer or the data registers should be read:
 **returnError = readRegister(&tempRegVal, KX13X_INC4);
  if( returnError != KX13X_SUCCESS )
    return false;

  if( tempRegVal & 0x40 ){ // If Buffer interrupt is enabled, then we'll read accelerometer data from buffer register.**

I currently use the TRIGGER mode for the sample buffer and put the sensor to sleep.  I have indicated that the "watermark" be halfway in the buffer (in high res this means 86 sample buffer, so midway is 43).  When the sensor is triggered I read the buffer and plot it.  The trigger values should be at sample 43 but it always show up at the beginning of the buffer.

There's something wrong with it but can't find the issue so far.

edspark, 2021-09-09:
I looked at this for about an hour but I'm a little stumped as well. I'll keep searching, but the code around setting is fine. Hm, a thought: If the board is asleep, then it would have to wake up and then trigger an interrupt at the set watermark. Is it possible your getting reading at the beginning of the buffer because your reading it as soon as it wakes up and not at the watermark?

ahermida69, 2021-09-09:
My understanding is that even if the sensor is asleep, it keeps reading values at the ODR that it was set.  The wake up engine samples at a much slower rate (100Hz max).  In other words the buffer is continually filled.  Once the trigger value has been measured that dataset is located at the watermark location.  In hi-res the buffer is 86 long so once the trigger occurs, 43 previous datasets are kept and the sensor continues reading values after the trigger value and storing them.  Once the buffer is full, the BFI is asserted and its time to read the buffer.

I think only ONCE was I able to see the trigger data right in the middle of the buffer, all other times it shows at the beginning of the buffer.  Note:  I always store the buffer in a file and plot it or use SerialPlot to watch it "realtime".  That's how I know.

It might be that the code works fine but the trigger buffer method is finicky and requires certain settings specifically and/or it might work for every case.  Not sure...

ahermida69, 2021-09-09:
[image]
[image]

HabermannR, 2021-12-11:
**returnError = readRegister(&tempRegVal, KX13X_INC4);
Why use KX13X_INC4 here?
KX13X_BUF_CNTL2 makes more sense:
BUFE – controls activation of the sample buffer.
BUFE = 0 – sample buffer inactive
BUFE = 1 – sample buffer active

And then it should be 
if( tempRegVal & 0x80 )
???

HabermannR, 2021-12-11:
or returnError = readRegister(&tempRegVal, KX13X_CNTL1);
if( tempRegVal & 0x20 )

DRDYE – Data Ready Engine enable bit.
DRDYE = 0 – Data Ready Engine is disabled
DRDYE = 1 – Data Ready Engine is enabled

ahermida69, 2021-12-14:
@HabermammR at some point I decided not to use the buffer read and actually develop my own routine since I was having mixed results and I needed some additional functions.

HabermannR, 2021-12-14:
@ahermida69 
Yes, same here. Do you want to share? I can upload my code on the weekend and invite you to the repository.

edspark, 2022-10-12:
The library has gotten a massive overhaul so I'm going to close this as it references old parts of the library. If there is some interest in this topic please reopen and I'd be happy to give it another look.

closed 2022-10-12 by edspark