sparkfun/SparkFun_KX13X_Arduino_Library #10: Lots of compiler warnings and Faults on Teensy 4.0

KurtE, 2021-11-17:
A thread was created on the PJRC forum that said that when they configured the sketch to run on SPI, the sketch would fault.
More details in the thread: https://forum.pjrc.com/threads/68743-SPI-Problem-Teensy-4-1-with-SparkFun-Qwiic-KX13x-library?p=293579#post293579

I did verify that the sketch is faulting in the SPI library, which I will try to localize down:   

Thought I would take a look and built the sketch and find there are lots of compiler warnings when I compile this library:
```
QwiicKX13xCore::setRange(uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:136:17: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( range < 0 | range > 3)
                 ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::setOutputDataRate(uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:159:16: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( rate < 0 | rate > 15 )
                ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::setInterruptPin(bool, uint8_t, uint8_t, bool)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:200:44: warning: comparison of constant '4' with boolean expression is always false [-Wbool-compare]
   else if( latchControl < 0 | latchControl > 4 )
                                            ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:200:29: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   else if( latchControl < 0 | latchControl > 4 )
                             ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::routeHardwareInterrupt(uint8_t, uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:223:15: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( rdr < 0 | rdr > 128 )
               ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::setBufferThreshold(uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:290:17: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( threshold < 2 | threshold > 171 )
                 ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:303:15: warning: statement has no effect [-Wunused-value]
     threshold == 86; 
               ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::setBufferOperation(uint8_t, uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:319:22: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( resolution < 0 | resolution > 1 )
                      ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:321:25: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
   if( operationMode < 0 | operationMode > 2 )
                         ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'KX13X_STATUS_t QwiicKX13xCore::readMultipleRegisters(uint8_t, uint8_t*, int16_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:449:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for(size_t i = 1; i < numBytes; i++) {
                       ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:474:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for(size_t i = 0; i < numBytes; i++) {
                       ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'outputData QwiicKX132::getAccelData()':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:593:11: warning: unused variable 'tempRegVal' [-Wunused-variable]
   uint8_t tempRegVal;
           ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:594:18: warning: unused variable 'returnError' [-Wunused-variable]
   KX13X_STATUS_t returnError;
                  ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'outputData QwiicKX134::getAccelData()':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:682:11: warning: unused variable 'tempRegVal' [-Wunused-variable]
   uint8_t tempRegVal;
           ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:683:18: warning: unused variable 'returnError' [-Wunused-variable]
   KX13X_STATUS_t returnError;
                  ^
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp: In member function 'bool QwiicKX13xCore::initialize(uint8_t)':
C:\Users\kurte\Documents\Arduino\libraries\SparkFun_KX13X_Arduino_Library\src\SparkFun_Qwiic_KX13X.cpp:97:3: warning: 'returnError' may be used uninitialized in this function [-Wmaybe-uninitialized]
   if( returnError == KX13X_SUCCESS )
   ^
```

So far not sure if any of these are causing the fault, but some of them are  probably bugs: 

Example:
```
bool QwiicKX13xCore::setRange(uint8_t range){

  if( range < 0 | range > 3)
    return false;
```
The statement: if( range < 0 | range > 3)
Is probably wrong two different ways: 
I am assuming that what is meant is: f( range < 0 || range > 3)
And note: as range is defined as uint8_t which is unsigned, it will never be < 0.

There are several lines like the above.

Also in setInterruptPin there is the line:
```
  else if( latchControl < 0 | latchControl > 4 )
```
This has same issue, plus latchControl is defined as a bool ...

Will update this more after I see where the fault is happening.

KurtE, 2021-11-18:
It is faulting, because _spiPort is not set to anything so you are calling through a null pointer. 
My guess is that there needs to be something like: _spiPort = &spiPort;
at about line 38.  Something like:
```
uint8_t QwiicKX13xCore::beginSPICore(uint8_t CSPin, uint32_t spiPortSpeed, SPIClass &spiPort)
{
	_i2cPort = NULL;
	_spiPortSpeed = spiPortSpeed;
  _spiPort = &spiPort;
  ```

edspark, 2021-11-18:
Yep you're absolutely right. Thank you @KurtE. I'm updating, or am most of the way through updating, the library this morning.

edspark, 2021-11-18:
Thank you again for the issue @KurtE . I have fixed various typos and overall SPI communication with this [commit](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/commit/fcbfc39f3830d7c8ed5a17039e7ac86828eb41f1) and this [one](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/commit/6ddb18267b7e43ea310f3851195fcc98d381bb98). The latest release of 1.0.7 is the one that includes these fixes.

closed 2021-11-18 by edspark