sparkfun/SparkFun_KX13X_Arduino_Library #11: Is there a simple way to apply LPF here?

schosdas, 2022-02-24:
I would like to apply the LPF (Low Pass Filter) using sensor kx134 and this library. Is there any example code for this? 
Thank you.

ahermida69, 2022-03-08:
Here's a quick piece of code for you.  You'll need to modify it to your code:

```
// gx, gy, gz are the sensor values you wish to filter (inputs)
// gxf, gyf, gzf are the filtered values (outputs)

// previous sensor values, initialize with 0.0
gx_prev = 0.0, gy_prev = 0.0, gz_prev = 0.0; 

//sampling time in seconds,
//if reading at 1000Hz (samples per second) then dt = 1/1000 = 0.001
dt = 0.001;                                                     

//filter cutoff frequency assume 300Hz                                                                     
fc = 300.0;                                                    
Tf =   1.0/(2.0*PI*fc)
alpha = Tf/(Tf + dt);
beta = 1.0 - alpha;

LOOP {
  //read gx, gy and gz

  gxf = alpha*gx_prev + beta*gx;
  gyf = alpha*gy_prev + beta*gy;
  gzf = alpha*gz_prev + beta*gz;

  gx_prev = gxf
  gy_prev = gyf
  gz_prev = gzf

}
```

edspark, 2022-10-12:
This is a rather old issue, but have you looked at ODCNTL register (pg 25 of [technical reference](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/blob/main/Documentation/KX134/KX134-1211-Technical-Reference-Manual-Rev-4.0.pdf))? There are different low pass filter roll off settings there. If this is something that could be of some use - I'd gladly add it to the library. I'm going to close this issue as it is quite old, but if you're still interested please reopen.

closed 2022-10-12 by edspark