sparkfun/SparkFun_KX13X_Arduino_Library #12: Kx134 Data Quality

doludiran, 2022-03-08:


ahermida69, 2022-03-08:
If you leave the sensor stationary, what is the min/max/average for the 50Hz and for the 25600Hz readings?

doludiran, 2022-03-08:
Thank you for your quick response. I took a 100 samples and here's the results

25600Hz:
                     X-Axis    Y-Axis    Z-Axis
Min:             -0.327     -0.35     -0.679
Max:            0.4089    0.3906  1.4
Average:     0.0297    0.0158    1.0206

50Hz:
                     X-Axis    Y-Axis    Z-Axis
Min:             0.0266   0.0117    1.0107
Max:            0.0359    0.0256     1.02
Average:     0.0311    0.01835   1.0145

It seems that averaging makes it better. My concern is that, this will affect frequency spectrum data. If I set the ODR to 25600Hz and I average the points say 100 samples to 1 point, when plotting the FFT for the samples, what will be the sampling rate for that data set?

ahermida69, 2022-03-09:
What is your g setting? The higher the g setting and sampling rate the more noise you will get.  My suggestion is to run the code multiple times with various ODR values and # of samples to average.  Look at the FFT for each case and the impact on the results.  I don't know your exact requirements or the application so its difficult to suggest more things.

doludiran, 2022-03-09:
Turns out averaging will not work for my application. I want to be able to sample at 13kHz and above and it is not possible to achieve this with averaging the points. Is there any way to achieve this with less noise? Atleast to get values closer to the 50Hz results?

edspark, 2022-10-12:
If you do have other questions please reopen a new issue!

closed 2022-10-12 by edspark