sparkfun/SparkFun_KX13X_Arduino_Library #15: Documentation and example unclear on Range

youcangetme, 2022-12-26:
I have spent the past hour reading the Sparkfun docs and I can't seem to nail down where the values below are defined for setting the range (https://learn.sparkfun.com/tutorials/triple-axis-accelerometer-breakout---kx13x-qwiic-hookup-guide-/kx13x-arduino-library);

- KX132_RANGE2G
- KX132_RANGE4G
- KX132_RANGE8G
- KX132_RANGE16G
- KX134_RANGE8G
- KX134_RANGE16G
- KX134_RANGE32G
- KX134_RANGE64G

I was able to find these in the headers;

```
#define SFE_KX132_RANGE2G  0x00
#define SFE_KX132_RANGE4G  0x01
#define SFE_KX132_RANGE8G  0x02
#define SFE_KX132_RANGE16G 0x03
#define SFE_KX134_RANGE8G  0x00
#define SFE_KX134_RANGE16G 0x01
#define SFE_KX134_RANGE32G 0x02
#define SFE_KX134_RANGE64G 0x03
```

Funny part is in the examples the range is defined and apparently C&P the same everywhere;

`kxAccel.setRange(0x18);        //  16g Range`

If you notice, the `SFE_KX134_RANGE16G 0x01` value is NOT `0x18`.

Maybe you all could shed some light on the matter and get the docs a bit more accurate on this topic. I am specifically looking for the 8G value.

youcangetme, 2022-12-26:
I was able to locate the usage for these values;

````
  switch( range ) {
    case SFE_KX134_RANGE8G:
....
````

So is it safe to assume that the examples are incorrect (C&P bug), the docs have a typo and the range is the `#define`s?

VancouverUmbrella, 2023-01-07:
I encountered exactly the same issues in the examples and documentation. And like @youcangetme, I also found the apparently conflicting info in the headers.

I understand that the KX13xx Arduino library was updated in October 2022 and that the online tutorials and examples may now be 'out of sync'. More clarity and consistency would definitely make life easier.

youcangetme, 2023-01-07:
I used the #define and that works. It's really just a int from 1 to 3 if I remember correctly. The code looks kind of rushed and using hex when an int would suffice isn't helpful.

edspark, 2023-01-11:
Thanks for filing this issue, it seems I missed some things when I did the large overhaul. I'm looking at this now. Also: there is a internal note for our docs team to update the tutorial.