sparkfun/SparkFun_KX13X_Arduino_Library #17: Arduino library, example1 sketch, dataReady() always returns false

VancouverUmbrella, 2023-01-07:
Hey there,

I've run into a blocker that has me spinning my wheels. Specifically, in the Sparkfun KX13X Arduino Library [https://github.com/sparkfun/SparkFun_KX ... no_Library](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library), in **example1_basic-readings**, the check for dataReady() always returns false. This is the case even though the accelerometer does actually return plausible data when I comment out the check for dataReady.

The relevant code is
`ln 63 kxAccel.enableDataEngine();     // Enables the bit that indicates data is ready.`

and
`ln 73 if( kxAccel.dataReady() )`

A SFE tech has confirmed that the problem is reproducible:
https://forum.sparkfun.com/viewtopic.php?f=105&t=58836&p=238164#p238164

Kind regards,
Andrew

edspark, 2023-01-11:
Yep, confirmed that Teensy has different behavior. Let me dive into why that would be the case.

edspark, 2023-01-11:
Alright sort of an unexpected but expected problem with the Teensy. It turns out it's too fast! What's happening is that I wasn't giving the proper 2ms delay after resetting the accelerometer: `softwareReset()`.  Here's an image of the accelerometer running without a delay on the Artemis Micromod processor: 

[image]

Here's the same code running on a Teensy:

[image]

The relevant parts are the "retVal" which should be a zero indicating success but are instead a -1. Give me a few minutes to remove my debug messages and add the delay to all the examples.

edspark, 2023-01-11:
No, thank you - fix is in place and the new release is out.

closed 2023-01-11 by edspark