sparkfun/SparkFun_KX13X_Arduino_Library #23: SPI higher that 2MHz

AnasHimmi, 2023-07-01:
Hi there,

According to the datasheet of the KX134-1211, it is possible to use an SPI clock up to 10MHz, however, when try too go over 2MHz, the accelerometer doesn't initialize. For example : 
`kxAccel.begin(SPI,SPISettings(3000000,MSBFIRST,SPI_MODE0),chip)`
doesn't work, and when setting a higher clock only for reading the acceleration, i get wrong data.
Do you think it is doable to have higher clock frequency?

Best regards,
Anas

ahermida69, 2023-11-15:
Did you solve this issue? What type of processor are you using?

delijati, 2024-05-06:
Would also be interested in this. @AnasHimmi can you provide an example Program so i can try it too.

edspark, 2024-07-08:
Sorry for the late response, I'll need a bit more information. What MCU are you using, what's your code look like? I'm looking at data output at 10MHz here on my [ESP32 C3](https://www.sparkfun.com/products/23484):

[image]

This screenshot is from my Salea logic analyzer. If you look closely enough on the "SCLK" bar, you'll see that it is indeed running at 10MHz and I found the data to be very similar when reading data at 2MHz. 

Let me know,
Thanks!

edspark, 2024-07-23:
Please reopen this issue if you find that it's not reaching the speeds you need. 

Thanks!

closed 2024-07-23 by edspark

delijati, 2024-07-23:
> Sorry for the late response, I'll need a bit more information. What MCU are you using, what's your code look like? I'm looking at data output at 10MHz here on my [ESP32 C3](https://www.sparkfun.com/products/23484):
> 
> ![10MHz](https://private-user-images.githubusercontent.com/18684691/346728147-ea38e5db-41a7-4e82-936a-0e34bad4b697.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE3NTE2NzQsIm5iZiI6MTcyMTc1MTM3NCwicGF0aCI6Ii8xODY4NDY5MS8zNDY3MjgxNDctZWEzOGU1ZGItNDFhNy00ZTgyLTkzNmEtMGUzNGJhZDRiNjk3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIzVDE2MTYxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0YjFkNTE5ODc5M2JlN2RlYmJlMTRlMjM4MThiMWRlNTlhNDczMTVmMDdlM2I3NmIyZTBhMGYzMjMyNzJjYjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.06BmXWgAsEmPweIbS8IyXoATPaq9SSgbtHX0aMj6e_g)
> 
> This screenshot is from my Salea logic analyzer. If you look closely enough on the "SCLK" bar, you'll see that it is indeed running at 10MHz and I found the data to be very similar when reading data at 2MHz.
> 
> Let me know, Thanks!

Can you add here your version as an reference