sparkfun/SparkFun_KX13X_Arduino_Library #25: Explanation of I2C Write

tay10r, 2025-01-09:
I'm curious about this line:

https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library/blob/c4a4b912a08b86e7189469dba79661cba8039298/src/SparkFun_Qwiic_KX13X.cpp#L127

As far as I can tell from the manual, there isn't a register mapped at this address and the software reset is done by setting a bit in the CNTL 2 register (which happens shortly after this line). What is this line of code doing?

edspark, 2025-01-09:
@PaulZC Can you give some insight here - looks like you added it a while back. I can confirm what @tay10r mentions, that there isn't anything in the datasheet regarding that particular value.

PaulZC, 2025-01-09:
More detail on page 4 here:

https://fscdn.rohm.com/kionix/en/document/TN027-Power-On-Procedure.pdf

@tay10r : please close this issue if this answers your question.

Best wishes,
Paul

tay10r, 2025-01-09:
As you mentioned, it's all explained on page 4. Thanks for sharing and for the quick response!

closed 2025-01-09 by tay10r